Octanis1/Octanis1-Mainboard-Firmware_MSP_EXP432P401RLP #6: I2C hangs and blocks all tasks depending on it

bitmorse, 2016-05-27:
BNO055 I2C communication can cause problems in that it answers too late or because of other issues. The result is that the I2C driver of TI-RTOS gets blocked forever and as a result all other tasks using I2C. Without BNO055 requests, bug does not appear.

bitmorse, 2016-06-04:
"Without BNO055 requests, bug does not appear." - is this still true?

traffael, 2016-07-05:
Yes, still true!

Failed workarounds:
- call I2C_close and I2C_open (--> segfault)
- toggle BNO055 reset pin

Not yet tried possible workarounds: 
- try IMU task restart
- try to toggle mainboard reset from EPS (instead of power cycling it. avoids IMU calibration status to be reset to 0 --> ensure no IMU_init is called after this!)
-

traffael, 2016-07-17:
-- abandoned. receive IMU data from sbc